rgrr/yapicoprobe #38: cmake fails

JustAnother1, 2023-04-23:
I tried to compile but cmake fails:
( My guess is that ${PROJECT} is not defined )
Could there be a how to build this section somewhere in the documentation?

<code>

CMake Error at CMakeLists.txt:23 (project):
  project PROJECT called with incorrect number of arguments

CMake Error at CMakeLists.txt:67 (add_executable):
  The target name "src/cdc_uart.c" is reserved or not valid for certain CMake
  features, such as generator expressions, and may result in undefined
  behavior.

CMake Error at CMakeLists.txt:87 (target_sources):
  Cannot specify sources for target "PRIVATE" which is not built by this
  project.

CMake Error at CMakeLists.txt:94 (target_include_directories):
  Cannot specify include directories for target "PRIVATE" which is not built
  by this project.

CMake Error at CMakeLists.txt:104 (target_sources):
  Cannot specify sources for target "PRIVATE" which is not built by this
  project.

CMake Error at CMakeLists.txt:122 (target_include_directories):
  Cannot specify include directories for target "PRIVATE" which is not built
  by this project.

CMake Error at CMakeLists.txt:151 (target_sources):
  Cannot specify sources for target "PRIVATE" which is not built by this
  project.

CMake Error at CMakeLists.txt:157 (target_link_libraries):
  Cannot specify link libraries for target "PRIVATE" which is not built by
  this project.

</code>

[CMakeError.log](https://github.com/rgrr/yapicoprobe/files/11304662/CMakeError.log)
[CMakeOutput.log](https://github.com/rgrr/yapicoprobe/files/11304663/CMakeOutput.log)

rgrr, 2023-04-23:
You are right: a "how to build" is missing.  In the meantime: how exactly do you try to build?

In the repo root there is a "stub" Makefile, which can be used.  `make cmake-create-debug` calls cmake with the correct parameters and `make all` compiles.  `make flash` flashes the probe if you are under Linux and have the udev rules installed.

JustAnother1, 2023-04-23:
<code>
git clone https://github.com/ . . .
cd yapicoprobe
git submodule update --init
mkdir build
cd build
export PICO_SDK_PATH=. . . /pico-sdk
cmake ..
</code>

A <code>make cmake-create-debug</code> + <code>make all</code> worked.

rgrr, 2023-04-24:
great.  I will add this to the documentation.

JustAnother1, 2023-04-24:
Is there a good reason to use Make + Ninja + CMake ?

rgrr, 2023-04-24:
make for Eclipse integration, ninja because it is fast, CMake because it generates ninja.

Question should be more why ninja and make?  Answer: because they exist.

Of course you can easily switch to just "make & cmake".

closed 2023-05-01 by rgrr