rgrr/yapicoprobe #48: pyocd i.MX RT1010 connecting with (ya)picoprobe: Unexpected ACK value (5)

IvoBCD, 2023-05-11:
When using (ya)picoprobe to program an NXP i.MX RT1010EVK, I get an "Unexpected ACK value (5)" with yapicoprobe v1.14.

Instead using LPC-Link2 (with DAPLink FW), JLink Pro (Segger FW) as SWD probes for pyocd (v0.35.0) works fine.

```console
# pyocd flash -vv  -t mimxrt1010 blinky-mimxrt1010_evk-3.3.0.elf 
0004387 D Project directory: /home/root/nxp/mimxrt1010 [session]
0007796 D Project directory: /home/root/nxp/mimxrt1010 [session]
0007798 D Loading user script: /home/root/nxp/mimxrt1010/pyocd_user.py [session]
0007827 D CMSIS-DAP v2 probe E6609CB2D36A7B29: protocol version 2.0.0 [dap_access_cmsis_dap]
0007856 I Target type is mimxrt1010 [board]
<MemoryMap@0x7f90ae6f50 regions=[<RamRegion@0x7f91023590 name=itcm type=MemoryType.RAM start=0x0 end=0x7fff length=0x8000 access=rwx>, <RomRegion@0x7f8f186d90 name=romcp type=MemoryType.ROM start=0x200000 end=0x217fff length=0x18000 access=rx>, <RamRegion@0x7f8f196710 name=dtcm type=MemoryType.RAM start=0x20000000 end=0x20007fff length=0x8000 access=rwx>, <RamRegion@0x7f8f196750 name=ocram type=MemoryType.RAM start=0x20200000 end=0x2020ffff length=0x10000 access=rwx>, <FlashRegion@0x7f8f197250 name=flexspi type=MemoryType.FLASH start=0x60000000 end=0x60ffffff length=0x1000000 access=rx blocksize=0x1000>]>
0007884 D Running task load_svd [sequencer]
0007887 D Running task pre_connect [sequencer]
0007887 D Running task dp_init [sequencer]
0007888 D Running task lock_probe [sequencer]
0007889 D Running task get_probe_capabilities [sequencer]
0007889 D Running task connect [sequencer]
0007928 D Default wire protocol selected; using SWD [dap]
0007930 D Sending deprecated SWJ sequence to select SWD [swj]
0007962 D DP IDCODE read failed; resending SWJ sequence (use dormant=False) [dap]
0007981 D Sending deprecated SWJ sequence to select SWD [swj]
0008013 D DP IDCODE read failed; resending SWJ sequence (use dormant=False) [dap]
0008032 D Sending SWJ sequence to select SWD; using dormant state [swj]
0008082 D DP IDCODE read failed; resending SWJ sequence (use dormant=False) [dap]
0008084 D Sending SWJ sequence to select SWD; using dormant state [swj]
0008145 D DP IDCODE read failed; resending SWJ sequence (use dormant=False) [dap]
0008146 D uninit session <pyocd.core.session.Session object at 0x7f8f187310> [session]
0008162 C Unexpected ACK value (5) returned by probe [__main__]
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 541, in read_dp_result_callback
    value = result()
            ^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1071, in read_reg_cb
    res = transfer.get_result()
          ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 157, in get_result
    self.daplink.flush()
  File "/usr/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 881, in flush
    self._read_packet()
  File "/usr/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1157, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 478, in decode_data
    data = self._decode_transfer_block_data(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 447, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/usr/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 370, in _check_response
    raise DAPAccessIntf.TransferError("Unexpected ACK value (%d) returned by probe" % ack)
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: Unexpected ACK value (5) returned by probe

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/subcommands/load_cmd.py", line 100, in invoke
    with session:
  File "/usr/lib/python3.11/site-packages/pyocd/core/session.py", line 402, in __enter__
    self.open()
  File "/usr/lib/python3.11/site-packages/pyocd/core/session.py", line 540, in open
    self._board.init()
  File "/usr/lib/python3.11/site-packages/pyocd/board/board.py", line 142, in init
    self.target.init()
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/coresight_target.py", line 123, in init
    super().init()
  File "/usr/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 172, in init
    seq.invoke()
  File "/usr/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/usr/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 466, in _connect
    connector.connect()
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 256, in connect
    self._idr = self.read_idr()
                ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 285, in read_idr
    dpidr = self._dp.read_dp(DP_IDR, now=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 822, in read_dp
    return read_dp_cb()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 810, in read_dp_cb
    result = result_cb()
             ^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 546, in read_dp_result_callback
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe
```

rgrr, 2023-05-11:
Hello Ivo,
why do you think, that this is a problem of the probe?  Does your hardware work with another probe and with pyocd?  Which version of pyocd do you use?

IvoBCD, 2023-05-12:
```console
# pyocd -V
0.35.0
```

@rgrr The very latest pyocd v0.35.0, my hardware does indeed work with other probes:

```console
# pyocd list
  #   Probe/Board                                     Unique ID   Target  
--------------------------------------------------------------------------
  0   NXP Semiconductors LPC-LINK2 CMSIS-DAP V5.361   NRAQBQHR    n/a     
# # pyocd flash -v -t mimxrt1010 blinky-mimxrt1010_evk-3.3.0.elf 
0007895 I Target type is mimxrt1010 [board]
0008091 I DP IDR = 0x0bd11477 (v1 MINDP rev0) [dap]
0008263 I AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0) [discovery]
0008330 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e:NXP part=88c) [rom_table]
0008366 I [0]<e00fe000:ROM class=1 designer=43b:Arm part=4c8> [rom_table]
0008367 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b:Arm part=4c8) [rom_table]
0008398 I   [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c7> [rom_table]
0008399 I     AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b:Arm part=4c7) [rom_table]
0008434 I     [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0008449 I     [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0008603 I     [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=00e> [rom_table]
0008617 I     [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0008645 I   [1]<e0041000:ETM M7 class=9 designer=43b:Arm part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0008678 I   [2]<e0042000:CTI CS-400 class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0008710 I [1]<e0040000:TPIU M7 class=9 designer=43b:Arm part=9a9 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0008726 I [2]<e0043000:TSGEN class=15 designer=43b:Arm part=101> [rom_table]
0008745 I IMXRT Boot Mode: Internal Boot [target_imxrt]
0008781 I CPU core #0 is Cortex-M7 r1p2 [cortex_m]
0008816 I FPU present: FPv5-SP-D16-M [cortex_m]
0008875 I 4 hardware watchpoints [dwt]
0008894 I 8 hardware breakpoints, 1 literal comparators [fpb]
0008988 I Loading /home/root/nxp/mimxrt1010/blinky-mimxrt1010_evk-3.3.0.elf [load_cmd]
0009095 I IMXRT Boot Mode: Internal Boot [target_imxrt]
0009111 I IMXRT Boot Device: 0 [target_imxrt]
[==================================================] 100%
0020870 I IMXRT Boot Mode: Internal Boot [target_imxrt]
0020875 I IMXRT Boot Device: 0 [target_imxrt]
0021016 I Erased 20480 bytes (5 sectors), programmed 18688 bytes (73 pages), skipped 4096 bytes (16 pages) at 1.86 kB/s [loader]
# 
```

and also with an old Segger JLink probe:

```console
# pyocd list
  #   Probe/Board          Unique ID   Target  
-----------------------------------------------
  0   Segger J-Link PLUS   609300812   n/a     
root@raspberrypi4-64:~/nxp/mimxrt1010# pyocd flash -v -t mimxrt1010 blinky-mimxrt1010_evk-3.3.0.elf 
0010792 I Target type is mimxrt1010 [board]
<MemoryMap@0x7f9b917890 regions=[<RamRegion@0x7f9b917350 name=itcm type=MemoryType.RAM start=0x0 end=0x7fff length=0x8000 access=rwx>, <RomRegion@0x7f9b9170d0 name=romcp type=MemoryType.ROM start=0x200000 end=0x217fff length=0x18000 access=rx>, <RamRegion@0x7f9b917610 name=dtcm type=MemoryType.RAM start=0x20000000 end=0x20007fff length=0x8000 access=rwx>, <RamRegion@0x7f9b9176d0 name=ocram type=MemoryType.RAM start=0x20200000 end=0x2020ffff length=0x10000 access=rwx>, <FlashRegion@0x7f9b9177d0 name=flexspi type=MemoryType.FLASH start=0x60000000 end=0x60ffffff length=0x1000000 access=rx blocksize=0x1000>]>
0012411 I DP IDR = 0x0bd11477 (v1 MINDP rev0) [dap]
0012927 I AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0) [discovery]
0013214 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e:NXP part=88c) [rom_table]
0013909 I [0]<e00fe000:ROM class=1 designer=43b:Arm part=4c8> [rom_table]
0013916 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b:Arm part=4c8) [rom_table]
0015037 I   [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c7> [rom_table]
0015043 I     AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b:Arm part=4c7) [rom_table]
0016163 I     [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0016173 I     [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0016182 I     [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=00e> [rom_table]
0016192 I     [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0016205 I   [1]<e0041000:ETM M7 class=9 designer=43b:Arm part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0016219 I   [2]<e0042000:CTI CS-400 class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0016232 I [1]<e0040000:TPIU M7 class=9 designer=43b:Arm part=9a9 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0016241 I [2]<e0043000:TSGEN class=15 designer=43b:Arm part=101> [rom_table]
0016247 I IMXRT Boot Mode: Internal Boot [target_imxrt]
0016252 I CPU core #0 is Cortex-M7 r1p2 [cortex_m]
0016256 I FPU present: FPv5-SP-D16-M [cortex_m]
0016267 I 4 hardware watchpoints [dwt]
0016276 I 8 hardware breakpoints, 1 literal comparators [fpb]
0016303 I Loading /home/root/nxp/mimxrt1010/blinky-mimxrt1010_evk-3.3.0.elf [load_cmd]
0016335 I IMXRT Boot Mode: Internal Boot [target_imxrt]
0016337 I IMXRT Boot Device: 0 [target_imxrt]
[==================================================] 100%
0026757 I IMXRT Boot Mode: Internal Boot [target_imxrt]
0026759 I IMXRT Boot Device: 0 [target_imxrt]
0026882 I Erased 24576 bytes (6 sectors), programmed 22784 bytes (89 pages), skipped 0 bytes (0 pages) at 2.11 kB/s [loader]
# 
```

IvoBCD, 2023-05-12:
I suspect the problem is https://github.com/rgrr/yapicoprobe/issues/39

IvoBCD, 2023-05-12:
I saw this with yapicoprobe-0114-pico-0bf7eee.uf2 and picoprobe_cmsis v1.0.2 Pico firmware.
Having issues reproducing with yapicoprobe FW as I mostly run into RPi4 USB host controller issue:

```console
[145641.902293] usb 1-1.1: new full-speed USB device number 5 using xhci_hcd
[145642.019939] usb 1-1.1: New USB device found, idVendor=2e8a, idProduct=000c, bcdDevice= 1.14
[145642.028560] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[145642.037662] usb 1-1.1: Product: YAPicoprobe CMSIS-DAP
[145642.042944] usb 1-1.1: Manufacturer: RaspberryPi
[145642.047755] usb 1-1.1: SerialNumber: E6609CB2D36A7B29
[145642.054107] usb 1-1.1: Not enough bandwidth for new device state.
[145642.060622] usb 1-1.1: can't set config #1, error -28
```

rgrr, 2023-05-12:
Hmmm.... now I'm a little bit confused...

The above seems to be a different issue!?

IvoBCD, 2023-05-12:
Sorry about the confusion - replacing a USB cable fixed this. So, trying again at 4 kHz:

```console
+ pyocd list
  #   Probe/Board                         Unique ID          Target  
---------------------------------------------------------------------
  0   RaspberryPi YAPicoprobe CMSIS-DAP   E6609CB2D36A7B29   n/a     
+ pyocd flash -v -t mimxrt1010 -f 4khz blinky-mimxrt1010_evk-3.3.0.elf
0007996 I Target type is mimxrt1010 [board]
<MemoryMap@0x7f9d293590 regions=[<RamRegion@0x7f9d292350 name=itcm type=MemoryType.RAM start=0x0 end=0x7fff length=0x8000 access=rwx>, <RomRegion@0x7f9d292bd0 name=romcp type=MemoryType.ROM start=0x200000 end=0x217fff length=0x18000 access=rx>, <RamRegion@0x7f9d2935d0 name=dtcm type=MemoryType.RAM start=0x20000000 end=0x20007fff length=0x8000 access=rwx>, <RamRegion@0x7f9d293690 name=ocram type=MemoryType.RAM start=0x20200000 end=0x2020ffff length=0x10000 access=rwx>, <FlashRegion@0x7f9d293410 name=flexspi type=MemoryType.FLASH start=0x60000000 end=0x60ffffff length=0x1000000 access=rx blocksize=0x1000>]>
0008731 C Unexpected ACK value (5) returned by probe [__main__]
```

And at 1kHz, with debug logging:
```console
+ pyocd list
  #   Probe/Board                         Unique ID          Target  
---------------------------------------------------------------------
  0   RaspberryPi YAPicoprobe CMSIS-DAP   E6609CB2D36A7B29   n/a     
+ pyocd flash -v -L '*=debug' -t mimxrt1010 -f 1khz blinky-mimxrt1010_evk-3.3.0.elf
0004413 D Project directory: /home/root/nxp/mimxrt1010 [session]
0004414 D [cmd:0] New _Command [dap_access_cmsis_dap]
0007849 D There is no PEMICRO library. [pemicro]
0007851 D Not sending notification because no subscribers: event=target_override [notification]
0007851 D Not sending notification because no subscribers: event=frequency [notification]
0007852 D Project directory: /home/root/nxp/mimxrt1010 [session]
0007853 D Not sending notification because no subscribers: event=debug.traceback [notification]
0007855 D Loading user script: /home/root/nxp/mimxrt1010/pyocd_user.py [session]
0007858 D Not sending notification because no subscribers: event=debug.log_flm_info [notification]
0007858 D Not sending notification because no subscribers: event=allow_no_cores [notification]
0007883 D   USB OUT> (2) 00 fe [pyusb_v2_backend]
0007886 D   USB IN < (3) 00 01 01 [pyusb_v2_backend]
0007887 D   USB RD < (3) 00 01 01 [pyusb_v2_backend]
0007887 D   USB OUT> (2) 00 04 [pyusb_v2_backend]
0007888 D   USB IN < (8) 00 06 32 2e 30 2e 30 00 [pyusb_v2_backend]
0007889 D   USB RD < (8) 00 06 32 2e 30 2e 30 00 [pyusb_v2_backend]
0007890 D CMSIS-DAP v2 probe E6609CB2D36A7B29: protocol version 2.0.0 [dap_access_cmsis_dap]
0007890 D   USB OUT> (2) 00 ff [pyusb_v2_backend]
0007891 D   USB IN < (4) 00 02 00 04 [pyusb_v2_backend]
0007892 D   USB RD < (4) 00 02 00 04 [pyusb_v2_backend]
0007892 D   USB OUT> (2) 00 f0 [pyusb_v2_backend]
0007894 D   USB IN < (3) 00 01 31 [pyusb_v2_backend]
0007894 D   USB RD < (3) 00 01 31 [pyusb_v2_backend]
0007895 D [cmd:1] New _Command [dap_access_cmsis_dap]
0007896 D [cmd:2] New _Command [dap_access_cmsis_dap]
0007921 I Target type is mimxrt1010 [board]
0007922 D trace: open [cmsis_dap_probe]
0007945 D [cmd:3] New _Command [dap_access_cmsis_dap]
0007946 D trace: set_clock(freq=1000) [cmsis_dap_probe]
0007947 D   USB OUT> (5) 11 e8 03 00 00 [pyusb_v2_backend]
0007949 D   USB IN < (2) 11 00 [pyusb_v2_backend]
0007950 D   USB RD < (2) 11 00 [pyusb_v2_backend]
<MemoryMap@0x7f9b394b90 regions=[<RamRegion@0x7f9cc3bd50 name=itcm type=MemoryType.RAM start=0x0 end=0x7fff length=0x8000 access=rwx>, <RomRegion@0x7f9b38a850 name=romcp type=MemoryType.ROM start=0x200000 end=0x217fff length=0x18000 access=rx>, <RamRegion@0x7f9b38ba10 name=dtcm type=MemoryType.RAM start=0x20000000 end=0x20007fff length=0x8000 access=rwx>, <RamRegion@0x7f9b394950 name=ocram type=MemoryType.RAM start=0x20200000 end=0x2020ffff length=0x10000 access=rwx>, <FlashRegion@0x7f9b394a50 name=flexspi type=MemoryType.FLASH start=0x60000000 end=0x60ffffff length=0x1000000 access=rx blocksize=0x1000>]>
0007953 D Running task load_svd [sequencer]
0007955 D Running task pre_connect [sequencer]
0007956 D Running task dp_init [sequencer]
0007956 D Running task lock_probe [sequencer]
0007957 D Running task get_probe_capabilities [sequencer]
0007958 D Running task connect [sequencer]
0007970 D trace: connect(DEFAULT) [cmsis_dap_probe]
0007971 D   USB OUT> (2) 02 01 [pyusb_v2_backend]
0007974 D   USB IN < (2) 02 01 [pyusb_v2_backend]
0007995 D   USB RD < (2) 02 01 [pyusb_v2_backend]
0007996 D   USB OUT> (5) 11 e8 03 00 00 [pyusb_v2_backend]
0007997 D   USB IN < (2) 11 00 [pyusb_v2_backend]
0007998 D   USB RD < (2) 11 00 [pyusb_v2_backend]
0008011 D   USB OUT> (6) 04 02 50 00 00 00 [pyusb_v2_backend]
0008013 D   USB IN < (2) 04 00 [pyusb_v2_backend]
0008014 D   USB RD < (2) 04 00 [pyusb_v2_backend]
0008027 D   USB OUT> (2) 13 00 [pyusb_v2_backend]
[... 207 lines cut ...]
    super().init()
  File "/usr/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 172, in init
    seq.invoke()
  File "/usr/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/usr/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 466, in _connect
    connector.connect()
[... cut ...]
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 822, in read_dp
    return read_dp_cb()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 810, in read_dp_cb
    result = result_cb()
             ^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 546, in read_dp_result_callback
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe

rgrr, 2023-05-12:
hmmm, different voltage levels?

I would try 500kHz at least.

rgrr, 2023-05-19:
Hello Ivo, any progress on this issue?

IvoBCD, 2023-05-19:
@rgrr I did check voltages (found it to be same 3.3V as on the other probes where it worked). Will retest with pyocd 0.35.1 shortly, having also swapped to a Pico with newer chiprev.

rgrr, 2023-05-19:
Thanks for the fast response.  Did not know, that there are different RP2040 revisions around.

rgrr, 2023-05-19:
> I suspect the problem is #39

You could give https://github.com/rgrr/yapicoprobe/releases/tag/rg-1.06 (up to 1.08) a try.  Those used CMSIS-DAP 2.1.2.

JustAnother1, 2023-07-02:
@rgrr The RP2040 has chiprevs B0 to B2 (until now) the RP2040 datasheet specifies the bugs that each newer revision fixed. The revision is also printed on the chip. The upper text line (right below the Raspberry logo read RP2-<revision> ...

rgrr, 2023-07-02:
> @rgrr The RP2040 has chiprevs B0 to B2 (until now) the RP2040 datasheet specifies the bugs that each newer revision fixed. The revision is also printed on the chip. The upper text line (right below the Raspberry logo read RP2- ...

@JustAnother1 Have you found any erratum which might affect PIO operation?

@IvoBCD any news on this issue?  Otherwise I will close it as "not reproducible"

JustAnother1, 2023-07-12:
The RP2040 Datasheet Appendix B Does not list an errata that relates to PIO.

rgrr, 2023-07-12:
cannot reproduce

rgrr, 2023-07-21:
Hello @IvoBCD ,

a problem similar to yours appeared with STM32.  The good thing is, that this has been resolved with #63.

If you are in the mood and have time it would be very kind if you could verify if this also helps with your issue.

I will later on provide some images.

closed 2024-06-27 by rgrr