rgrr/yapicoprobe #60: Project Fails to Build

tornupnegatives, 2023-07-19:
I am trying to compile yapicoprobe for the Raspberry Pi Pico, on the following build system:
- OS: Ubuntu 20.04.6 LTS x86_64
- CPU: Intel i7-8750H (12) @ 4.100GHz
- Memory: 16GB
- Toolchain: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
- Pico SDK: v1.5.1

I initiate the build with the following steps:
```shell
$ git clone https://github.com/rgrr/yapicoprobe
$ cd yapicoprobe
$ git submodule update --init --recursive
$ make cmake-create-release && make all -j
```

I am greeting with a slew of errors, which I have attached as a logfile below. Could anyone point me in the right direction?

Thanks,
Joseph

[make.log](https://github.com/rgrr/yapicoprobe/files/12099542/make.log)

rgrr, 2023-07-19:
Not sure if your arm-none-eabi-gcc is the culprit.  It seems to be very old.

Could you post the output of `make cmake-create-release`?

tornupnegatives, 2023-07-19:
I have upgraded to a more recent GCC ARM Embedded toolchain (arm-none-eabi-gcc (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)) 11.2.1 20220111) and repeated the steps listed above. Unfortunately, the error persists. I have attached logs which correspond to the `make cmake-create-release` and `make all -j` invocations. Unfortunately, the same errors remain.

[make-cmake-create-release.log](https://github.com/rgrr/yapicoprobe/files/12100364/make-cmake-create-release.log)

[make-all.log](https://github.com/rgrr/yapicoprobe/files/12100365/make-all.log)

rgrr, 2023-07-20:
My thoughts:

If you compare your and my cmake creation, you will see some differences at the top.

Are you sure that you have the SDK correctly installed?  Can you compile SDK examples?

make-all.log says something about gcc 9.4.0, the creation log 11.2.1.  Why?  And why don't you use anything newer?

It seems that you compiler doesn't like the `CFG_TUSB_MEM_SECTION CFG_TUSB_MEM_ALIGN tu_static const ntb_parameters_t ntb_parameters`.  Rewrite it as `static const ntb_parameters_t ntb_parameters`.  Same some lines below.

rgrr, 2023-07-20:
Nevertheless ncm_device from the original TinyUSB code uses the same construct (CFG_TUSB_MEM_SECTION CFG_TUSB_MEM_ALIGN tu_static const)

tornupnegatives, 2023-07-20:
## Preliminary

```shell
# Double-check build environment
$ make --version # 4.2.1
$ cmake --version # 3.25.1
$ arm-none-eabi-gcc --version # 12.2.1 20230214

# Grab fresh copy of Pico SDK
$ cd /opt && git clone https://github.com/raspberrypi/pico-sdk.git
$ cd pico-sdk && git submodule update --init --recursive

# Grab fresh copy of yapicoprobe
$ cd
$ git clone https://github.com/rgrr/yapicoprobe.git -b v1.15
$ cd yapicoprobe && git submodule update --init --recursive

# Build yapicoprobe
$ export PICO_SDK_PATH=/opt/pico-sdk
$ make cmake-create-release > make-cmake-create-release.log 2>&1
$ make all -j > make-all.log 2>&1
```

### Notes
> make-all.log says something about gcc 9.4.0, the creation log 11.2.1. Why? And why don't you use anything newer?

Here, the system reports that my GCC/G++ version is 9.4.0 for the _host_ platform, which will be used to build the `elf2uf2` and `pioasm` utilities. However, it will use the ARM GNU toolchain to build `yapicoprobe`.

## Fixes
At this point, our Cmake logs match closely, but I am still getting some of the same build errors.

> It seems that you compiler doesn't like the `CFG_TUSB_MEM_SECTION CFG_TUSB_MEM_ALIGN tu_static const ntb_parameters_t ntb_parameters`. Rewrite it as `static const ntb_parameters_t ntb_parameters`. Same some lines below.
> Perhaps `const` is the culprit...

Now, my compiler is unbothered by the `libusb` types. However, it does still complain about the `const` discrepancy. In this case, adding `const` where it is missing creates "assignment of read-only location" errors.

I use the below invocation to make the changes you suggest:
```shell
$ find . -type f -not -path '*/\.*' -exec sed -i 's/extern uint8_t tud_network_mac_address/extern const uint8_t tud_network_mac_address/g' {} +
$ sed -i 's/uint8_t tud_network_mac_address/const uint8_t tud_network_mac_address/g' src/get_config.c
$ make all -j > make-all-const-fix.log 2>&1
```

```
/home/joseph/yapicoprobe/src/get_config.c: In function 'get_config_init':
/home/joseph/yapicoprobe/src/get_config.c:59:32: error: assignment of read-only location 'tud_network_mac_address[0]'
   59 |     tud_network_mac_address[0] = 0xfe;     // 0xfe is allowed for local use, never use odd numbers here (group/multicast)
      |                                ^
/home/joseph/yapicoprobe/src/get_config.c:62:36: error: assignment of read-only location 'tud_network_mac_address[i]'
   62 |         tud_network_mac_address[i] = uID.id[i + (PICO_UNIQUE_BOARD_ID_SIZE_BYTES - sizeof(tud_network_mac_address))];
```
[make-cmake-create-release.log](https://github.com/rgrr/yapicoprobe/files/12112827/make-cmake-create-release.log)
[make-all.log](https://github.com/rgrr/yapicoprobe/files/12112825/make-all.log)
[make-all-const-fix.log](https://github.com/rgrr/yapicoprobe/files/12112826/make-all-const-fix.log)

rgrr, 2023-07-20:
Hmmm... obviously now the new compiler is working.  How about using unmodified sources?

tornupnegatives, 2023-07-20:
If I am only using the SWCLK and SWDIO pins, do I need to do anything with SWDIR and RESET, or may I leave them floating?

rgrr, 2023-07-21:
> If I am only using the SWCLK and SWDIO pins, do I need to do anything with SWDIR and RESET, or may I leave them floating?

Both are outputs, so there should be no problem leaving them open

rgrr, 2023-07-21:
After you can build images which does not work according to #61: what does the probes debug output says?

closed 2023-08-01 by rgrr