rgrr/yapicoprobe #67: Probe Locates RTT CB but CDC is Empty

tornupnegatives, 2023-07-26:
## Description
I have an STM32H7xx target configured as an RTT target which I would like to monitor via `yapicoprobe`. I have confirmed with a SEGGER JLINK cable that RTT works on the target, but I would strongly prefer not to be tied to propriety hardware...

The STM32H7xx target has 512 MB of SRAM, which would take too long to index via `yapicoprobe` so I have explicitly set the RTT CB address in `rtt_console.c`

```C
static uint32_t                 prev_rtt_cb = 0x2400b1c4;
```

When I connect `yapicoprobe` to the target, it finds the RTT CB, but the RTT CDC stream is empty!

```
(II) +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
(II) Target vendor     : Generic                                                
             0.039 ( 17) -
(II) Target part       : cortex_m.039 (  0) -
(II) Flash             : 0x00000000..0x000fffff (1024K)
(II) RAM               : 0x20000000..0x2003ffff (256K) 0.039 (  0) -
(II) SWD frequency     : 2000kHz                      0.039 (  0) -
(II) SWD max frequency : 10000kHz.039 (  0) -
(II) +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
(II) RTT_CB found at 0x2400b1c4                                                 
             0.040 (  1) -
(II) RTT_CB found at 0x2400b1c40.140 (100) -
(II) RTT_CB found at 0x2400b1c40.740 (600) -
(II) RTT_CB found at 0x2400b1c41.340 (600) -
(II) RTT_CB found at 0x2400b1c41.940 (600) -
(II) RTT_CB found at 0x2400b1c42.540 (600) -
(II) RTT_CB found at 0x2400b1c43.140 (600) -
...
```

Interestingly, when I try to read the RTT CB from my Nucleo board (STM327Hxx, again) which places the structure at address 0x200004dc, the RTT CDC is full of beautiful data. It seems that the issue has to do with the location of the buffer?

## Notes
One interesting observation, which you can see in the above sample of the debug CDC stream, is that `yapicoprobe` continuously reports the RTT CB at incrementing address locations. I am not sure if this is supposed to happen.

I also hypothesize that, although I have given `yapicprobe` the address of my RTT CB, because it exists outside of the range of RAM addresses specified in `pico.c`, it may not actually be able to interact with the structure.

```C
target_cfg_t target_device_generic = {
    ...
    .ram_regions[0].start           = 0x20000000,
    .ram_regions[0].end             = 0x20000000 + KB(256), // STM32H7xx: 0x20000000 + MB(512)
    ...
};
```

rgrr, 2023-07-26:
> One interesting observation, which you can see in the above sample of the debug CDC stream, is that yapicoprobe continuously reports the RTT CB at incrementing address locations. I am not sure if this is supposed to happen.

Haha, I guess this is related to your broken output with the carriage return at the wrong place.  What you see incrementing is the time.  Nevertheless I never saw repeating output of RTT_CB...

What irritates me is that your RTT_CB lies at 0x2400b1c4, but your RAM specification starts at 0x20000000.  So even with 512KByte RAM you will not reach that address.

Why don't you simply specify the correct RAM range in `src/daplink-pico/board/rp2040/pico.c` (target_device_generic) and leave the rest of the sources unmodified and see what the probe does?

tornupnegatives, 2023-07-26:
I'm not sure I understand. If ` .ram_regions[0].start = 0x20000000` and .ram_regions[0].end = 0x20000000 + MB(512) = 0x40000000`, it should be able to find the RTT CB. Right?

tornupnegatives, 2023-07-26:
Update:

If I configure the probe as such, I _do_ get RTT output.

```C
static uint32_t prev_rtt_cb = 0x2400b1c4;
```

```C
target_cfg_t target_device_generic = {
    ...
    .ram_regions[0].start           = 0x20000000,
    .ram_regions[0].end             = 0x20000000 + MB(512),
    ...
};
```

However, if I unset `prev_rtt_cb`, it never finds the RTT CB.

rgrr, 2023-07-26:
Far away from 0x24000000... 512KByte = 64KByte * 8 = 0x10000 * 8 = 0x80000 -> 0x20080000.

Check the RAM mapping of the device.

rgrr, 2023-07-26:
ok, checked the datasheet of the STM32H7 series.  It says that RAM is not consecutive.  And that is a problem...

The first 128KByte (DTCM)  are located from 0x20000000 .. + 128KByte, second ares starts at 0x24000000 (AXI-SRAM).  So there is a big gap in between.

Perhaps you can place the RTT buffer into the DTCM?

tornupnegatives, 2023-07-26:
> 512KByte

The configuration I am using places the end of RAM 512 _mega_ bytes past the start.

> The first 128KByte (DTCM) are located from 0x20000000 .. + 128KByte, second ares starts at 0x24000000 (AXI-SRAM). So there is a big gap in between.

Even if there is a large gap, if I have configured the probe to search between 0x20000000-0x3fffffff, it should eventually hit my target address, no? Would having gaps cause a failure?

> Perhaps you can place the RTT buffer into the DTCM?

Yes, that would be the optimal solution. Unfortunately, however, I am working with a large robotics firmware codebase where touching the linker script would be unwise...

tornupnegatives, 2023-07-26:
It does appear that the `target_cfg_t` struct intends to support R/W of multiple RAM banks; up to `MAX_REGIONS` different regions. Perhaps yapicoprobe may be configured with a new target for the STM327Hxx which knows of each of these banks?

```C
typedef struct __attribute__((__packed__)) target_cfg {
    ...
    region_info_t flash_regions[MAX_REGIONS];   /*!< Flash regions */
    region_info_t ram_regions[MAX_REGIONS];     /*!< RAM regions  */
    ...
} target_cfg_t;
```

rgrr, 2023-07-26:
I understand the problem.  But perhaps you could check, what segments are placed into the DTCM and put the RTT_CB there.  This would not involve changes to the linker script.  And the RTT_CB is small, the actual buffers could be somewhere else.  Your devops/architects should take debugging of the system into account.

Nevertheless, 0x0400_0000 means 64MByte... not sure how fast the probe scans memory (add some debug output!), but surely this takes some time t detect an RTT_CB at 0x2400_0000.

rgrr, 2023-07-26:
> It does appear that the target_cfg_t struct intends to support R/W of multiple RAM banks...

yes, the struct supports this.  But rtt_console does not.  Perhaps you can extend functionality!?

tornupnegatives, 2023-07-26:
> Your devops/architects should take debugging of the system into account

PREACH!

> not sure how fast the probe scans memory (add some debug output!), but surely this takes some time t detect an RTT_CB at 0x2400_0000

Yeah, I did some (entirely baseless) math assuming that if it takes 600ms/256KB, scanning through 512MB could take at least 20 min...

> Perhaps you can extend functionality!?

I will look into it :-)

tornupnegatives, 2023-08-09:
The new RTT control block search algorithm seems to work nicely for devices where the control block is in a "normal" location. It connects, reports on the CDC that it found the CB, and exposes a USB mass storage device with some useful files.

On my board where the RTT control block is outside of the reaches of memory, it spins away, and the USB mass storage device never appears. Of course, this is more of a "me" problem, since this issue is normally solvable via linker script. Although it would be interesting to know how SEGGER's RTT Viewer application manages to find the control block so quickly, regardless of where it resides...

```
570.215 (999) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
571.690 (999) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
573.165 (999) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
574.640 (999) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
576.115 (999) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
577.590 (999) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
```

TL;DR: It works ;-)

rgrr, 2023-08-09:
Thanks for the fast feedback.

I have changed the logic a little bit to prevent the loop on no rtt_cb.  Also there was a bug in the detection of generic target devices.  Perhaps you could try again?

I'm thinking about to make the RAM/Flash boundaries configurable for generic devices.

PS: Yes, would be interesting how Segger does it.

tornupnegatives, 2023-08-09:
No noticeable changes after pulling the latest `master`

rgrr, 2023-08-09:
Hmm, does the MSC appear?

tornupnegatives, 2023-08-09:
Not on the target whose RTT buffer cannot be located, no.

rgrr, 2023-08-09:
what kind of target is this exactly?  Any cheap dev board available?

tornupnegatives, 2023-08-09:
STM32H743

rgrr, 2023-08-10:
Which Nucleo dev board are you using?  For your experiments which do not show MSC: did you use an "official" image or did you compile yourself with your 512MB RAM size?

tornupnegatives, 2023-08-10:
I have a Raspberry Pi Pico flashed with the "official" yapicoprobe image provided in the GitHub releases. This, of course, does not search the entire 512 MB of RAM, but that does not mean it should necessarily block the MSC from appearing.

rgrr, 2023-08-10:
what does the debug output of the probe show?  I mean from the beginning.

tornupnegatives, 2023-08-10:
When using the `v1.18-pre` image to target the Nucleo board, I quite quickly get access to a USB CDC with debug output as well as the MSC.

On the PCBA, neither appear. It seems that the RTT CB search may be blocking other operations.

rgrr, 2023-08-10:
I'm wondering, how the debug interface on the MCU behaves if memory outside the actual ranges is accessed.

If I understand correctly, the DTCM RAM area starts at 0x2000_0000 and has 128K.  Perhaps if you limit the RTT search to 128K instead of extending it prevents the hang ups?

tornupnegatives, 2023-08-10:
Using the `v1.18-pre`, is the RTT search limit not 128K by default?

rgrr, 2023-08-10:
The probe tells it on its debug output:

```
[19:43:07.577871 0.000002] 0.003 (  3) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
[19:43:07.579592 0.001724] 0.004 (  1) - (II)                      Welcome to Yet Another Picoprobe v1.18-ce223df
[19:43:07.580828 0.001236] 0.004 (  0) - (II) Features:
[19:43:07.581249 0.000421] 0.004 (  0) - (II)   [CMSIS-DAPv1] [CMSIS-DAPv2] [MSC: DAPLink] [CDC: target] [CDC: probe debug] [Net-NCM SysView Echo IPerf]
[19:43:07.583065 0.001816] 0.004 (  0) - (II) Probe HW:
[19:43:07.583539 0.000474] 0.004 (  0) - (II)   Pico @ 240MHz
[19:43:07.584080 0.000542] 0.004 (  0) - (II) IP:
[19:43:07.584400 0.000321] 0.004 (  0) - (II)   192.168.14.1
[19:43:07.584892 0.000492] 0.004 (  0) - (II) Compiler:
[19:43:07.585295 0.000402] 0.004 (  0) - (II)   clang 18.0.0 - release build
[19:43:07.585995 0.000700] 0.004 (  0) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
[19:43:07.587489 0.001494] 0.017 ( 13) - (II) SWD clk req   : 10000kHz = 240000kHz / (6 * (4 + 0/256)), eff : 10000kHz
[19:43:07.588786 0.001298] 0.037 ( 20) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
[19:43:07.590296 0.001510] 0.038 (  1) - (II) Target vendor     : Generic
[19:43:07.590965 0.000669] 0.038 (  0) - (II) Target part       : cortex_m
[19:43:07.591629 0.000665] 0.038 (  0) - (II) Flash             : 0x00000000..0x000fffff (1024K)
[19:43:07.592610 0.000980] 0.038 (  0) - (II) RAM               : 0x20000000..0x2003ffff (256K)
[19:43:07.593563 0.000953] 0.038 (  0) - (II) SWD frequency     : 10000kHz
[19:43:07.594235 0.000672] 0.038 (  0) - (II) SWD max frequency : 10000kHz
[19:43:07.594910 0.000675] 0.038 (  0) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
[19:43:07.596398 0.001488] 0.137 ( 99) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
[19:43:07.597576 0.001178] 0.952 (815) - (II) ---- no RTT_CB found
...
```

So it assumes 256K RAM.

But there is some strange startup behaviour if no target is connected.  I have to check this.

rgrr, 2023-08-10:
Made some fixes in rtt_io and updated the images.  But I doubt that this will solve your issue.  I will have a look at it, but not today.

rgrr, 2023-08-20:
Could you check the latest release?  There you can setup r_start/r_end in runtime configuration to your needs

tornupnegatives, 2023-08-21:
What a cool approach! This does the trick. Thank you :)

rgrr, 2023-08-21:
Hey, great.  It was more or less a one-liner :-)

Im closing (closed by #88).

closed 2023-08-21 by rgrr