rgrr/yapicoprobe #100: Usage with probe-rs is not possible

scotty00X, 2023-11-05:
`v1.18` rsp. `v1.19-pre` (other version not tested) seems not work with the latest probe-rs https://github.com/probe-rs/probe-rs.

The error cause mentioned is mostly:
```
Error: Failed to open the debug probe.

Caused by:
    0: An error specific to a probe type occurred
    1: Error handling CMSIS-DAP command Info
    2: Unexpected answer to command
```
or

```

Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: An error specific to a probe type occurred
    2: Error handling CMSIS-DAP command Connect
    3: Command ID in response (:#02x) does not match sent command ID
```

Does anybody observes the same behavior? Should yapicoprobe work with `probe-rs`?

**To Reproduce**
Steps to reproduce the issue:

    RUST_LOG=probe_rs=trace probe-rs reset --chip RP2040

rgrr, 2023-11-05:
Hello Henrik

interesting.  I have never tested yapicoprobe with probe-rs because I'm currently not a rust guy, so I have no environment for testing.  May change over time.

Nevertheless: what are the actual requirements for probe-rs to connect to the target?  Which protocol is used?

scotty00X, 2023-11-05:
Hi Hardy,
thx for your reply. 

> It is mentioned as of version 0.10.0 this library can
> 
>     connect to a DAPLink, STLink or JLink ...

This is done via [cmsisdap](https://github.com/probe-rs/probe-rs/tree/master/probe-rs/src/probe/cmsisdap) protocol. I'm definitely not an expert on this topic.

At least the probe type, etc. is recognized as a  CMSIS-DAP probe:
```
RUST_LOG=probe_rs=trace probe-rs list
 INFO probe_rs: Writing log to "/home/hkuhn/.local/share/probe-rs/1699217905564.log"
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using libusb
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 0: YAPicoprobe CMSIS-DAP v2
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 1: YAPicoprobe CMSIS-DAP v1
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:     HID interface found
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 3, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 6, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 6, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 8, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: YAPicoprobe CMSIS-DAP: CMSIS-DAP device with 9 interfaces
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Will use interface number 1 for CMSIS-DAPv1
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using libusb, searching HID
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: CMSIS-DAP device with USB path: "/dev/hidraw4"
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:                 product_string: "YAPicoprobe CMSIS-DAP"
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:                      interface: 1
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Ignoring duplicate YAPicoprobe CMSIS-DAP (VID: 2e8a, PID: 000c, Serial: E6612483CB465D2A, CmsisDap)
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
The following debug probes were found:
[0]: YAPicoprobe CMSIS-DAP (VID: 2e8a, PID: 000c, Serial: E6612483CB465D2A, CmsisDap)
```

rgrr, 2023-11-06:
Ok, this seems to be reasonable.

What does it output if you are trying to connect to the target?  Do you know of probes working with the RP2040?

scotty00X, 2023-11-06:
Have a look [here](https://github.com/probe-rs/probe-rs/issues/1849#issuecomment-1793828247_) as an example what happens if I try to reset the target.

rgrr, 2023-11-07:
I will have a deeper look into this, but currently I'm short in time.  As I written in probe-rs, would be nice to have a more detailed log file.

BTW: is it possible to connect to other targets?  E.g. nRF52xxx?

plaes, 2023-12-07:
> I will have a deeper look into this, but currently I'm short in time. As I written in probe-rs, would be nice to have a more detailed log file.
> 
> BTW: is it possible to connect to other targets? E.g. nRF52xxx?

No, same issue with nrf528340. I added some debug logs to relevant probe-rs issue (both for yapicoprobe and working picoprobe).

rgrr, 2023-12-11:
...
> No, same issue with nrf52840. I added some debug logs to relevant probe-rs issue ([probe-rs/probe-rs#1849](https://github.com/probe-rs/probe-rs/issues/1849)) with logs for both yapicoprobe and picoprobe as well.

Thanks for the logs.  I will try to have a look at it during this week

rgrr, 2024-02-04:
Could you please give the latest release (1.20) a chance.  You have to change the configuration to rtt=0 for probe-rs to work.

rgrr, 2024-02-28:
probe-rs 0.23.0 solves this, see https://github.com/probe-rs/probe-rs/issues/2139

closed 2024-02-28 by rgrr

scotty00X, 2024-03-04:
The mentioned CMSIS-DAPv2 timeout fix in probe-rs establishes the interoperability with yapicoprobe.
thank you very much.