rgrr/yapicoprobe #101: Overclock issue with new RP2040

phb98, 2024-01-06:
I got a new patch of rp2040 (03-22). After flashing FW it does not boot at all( No LED flashing, no usb emulation). That FW is running fine on older RP2040 (04-20).

Rebuild the FW with lower CPU clock (PROBE_CPU_CLOCK_MHZ = 192MHz)  and the MCU boot just fine, try 216 Mhz and it becomes unstable(Random freeze).

Maybe considering a lower clock speed if it's does not affect performance much. I only use DAP and CDC and did not notice any difference between 2 clock speed yet(SWD was running at 20Khz)

rgrr, 2024-01-07:
Hello phb98

thanks for reporting this.  In an upcoming version I will lower standard CPU frequency to something more reasonable.

Don't know if you are aware of fine-tuning CPU freq: if the probe starts up, one can set CPU frequency via configuration: https://github.com/rgrr/yapicoprobe#runtime-configuration-of-probe-parameters

Concerning performance: there seems to be some bigger USB performance step between 192MHz and 216MHz.  Don't know why.

Do you know, if this is just a new batch or is it also a new chip revision?

phb98, 2024-01-11:
Hi
I have only tested 1 so far, so i can not confidentally if this apply to all new rp2040, maybe i'm just unlucky. But still running at twice default clock does not sound reliable to me. Last thing i want when debugging my project is debugging my debugger.

rgrr, 2024-02-28:
#105 closes this

closed 2024-02-28 by rgrr