rgrr/yapicoprobe #103: DAP_FingerprintTool (dap_util.c) returning E_DAPTOOL_UNKNOWN instead of correct tool

TGerrits-86, 2024-02-12:
I guess that the DAP_FingerprintTool  function return have a typo. 
From:
 `return (cnt **<** 3) ? E_DAPTOOL_UNKNOWN : tool;` 
It should be:
`return (cnt **>** 3) ? E_DAPTOOL_UNKNOWN : tool;`

rgrr, 2024-02-12:
No, I hope it is correct this way.

But I have to admit that the code is not very "clean".  "cnt" shows the number of samples put into the function.  If "cnt < 3", then the fingerprint is not complete and thus unknown.  If "cnt >= 3", then the fingerprint is complete and the probing result should be returned.

I have made some code changes to make this more "clean".

rgrr, 2024-02-12:
Question: do you see anything which does not work as expected, esp in debug console output?

TGerrits-86, 2024-02-12:
Indeed it's clearer, thanks for that. 
I'm debugging directly (GDB), no logs activated.

What I see is that when the I run "pyocd list" on the CLI, the "sample_no" is set to 0 (after the ++sample_no; it will go to "1") and request_len to 2. 

Entering the first "If(sample_no == 1)", E_DAPTOOL_PYOCD will be selected correctly but on the return sentence E_DAPTOOL_UNKNOWN will be selected since sample_no<3.

I'm using pyocd 0.36.0

(I m obtaining the same result using openocd, i.e., the first if selects the OpenOCD as expected, but the return will choose E_DAPTOOL_UNKNOWN)

rgrr, 2024-02-12:
This is expected behavior.  It needs three samples to have the complete fingerprint.  Beforehand "unknown" is returned.

Make "pyocd reset" and you will see which target configuration is actually used.  For me:

```
0.000 (...) - (II) =================================== DAPv2 connect target, host pyOCD with single big buffer, buffer: 1x1024bytes
0.008 (  8) - (II) SWD clk req   : 1000kHz = 120000kHz / (6 * (20 + 0/256)), eff : 1000kHz
1.328 (...) - (II) =================================== DAPv2 disconnect target
```

TGerrits-86, 2024-02-12:
You are completely right, I misunderstood the code. I reverted the changes on my side and it worked as you mentioned.
Sorry and Thanks for the good explanation.