rgrr/yapicoprobe #109: Error: CMSIS-DAP: SWD not supported

sbishop61, 2024-03-26:
I am trying to use a Pico with yapicoprobe to upload and debig another Pico.
Using VS Code, Platformio, with platform = raspberrypi and framework = arduino.
`debug_tool = cmsis-dap
upload_protocol = cmsis-dap
build_type = debug`

I get the following:
`Info : RP2040 Flash Bank Command
adapter speed: 5000 kHz

Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E46320165B3B1C21
Info : CMSIS-DAP: FW Version = 2.1.2
Info : CMSIS-DAP: Serial# = E46320165B3B1C21
Error: CMSIS-DAP: SWD not supported
Error: No Valid JTAG Interface Configured.
*** [upload] Error 4294967295`

Please excuse me if I have misunderstood, but I though SWD was supported.

rgrr, 2024-03-27:
In general this should work.  But I have never tested it with your environment above.

Questions:
- are you using the correct UF2 image?
- have you checked the cabling / voltage levels etc? https://github.com/ochamodev/raspberry_pico_setup_guide is a good starting point
- have you used another target?
- have you tried another probe firmware?  E.g. the mother of the pico probes: https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.0

sbishop61, 2024-03-27:
I should have said that I am using 2 Pico clones which have 16Mb flash. (W25Q32)

I am using yapicoprobe-0120-picow-f726bca.uf2
I have connected 3 wires as per the instructions. The target Pico is powered by USB

When I connect to the target, on the CLI UART I see:
`0.003 (  3) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
0.004 (  1) - (II)                      Welcome to Yet Another Picoprobe v1.20-f726bca
0.004 (  0) - (II) Features:
0.004 (  0) - (II)   [CMSIS: DAPv1] [CMSIS: DAPv2] [MSC: DAPLink] [CDC: target] [CDC: probe debug] [Net-NCM: SysView Echo IPerf]
0.004 (  0) - (II) Probe HW:
0.004 (  0) - (II)   Pico @ 240MHz (dual core)
0.004 (  0) - (II) IP:
0.004 (  0) - (II)   192.168.14.1
0.004 (  0) - (II) Compiler:
0.004 (  0) - (II)   clang 17.0.1 - release build
0.004 (  0) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
0.000 (...) - (II) FLASH: Copying custom flash code to 0x20010000 (620 bytes)
0.015 ( 15) - (II) SWD clk req   : 15000kHz = 240000kHz / (6 * (2 + 171/256)), eff : 14992kHz
0.025 ( 10) - (II)
0.026 (  1) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
0.026 (  0) - (II) Target vendor     : RaspberryPi
0.026 (  0) - (II) Target part       : RP2040
0.026 (  0) - (II) Flash             : 0x10000000..0x101fffff (2048K)
0.026 (  0) - (II) RAM               : 0x20000000..0x2003ffff (256K)
0.027 (  1) - (II) SWD frequency     : 15000kHz
0.027 (  0) - (II) SWD max frequency : 25000kHz
0.027 (  0) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
0.027 (  0) - (II)
0.127 (100) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
0.712 (585) - (II) ---- no RTT_CB found`

### Using Pi UF2: debugprobe_on_pico.uf2 I get different errors:
`Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
adapter speed: 5000 kHz

Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E46320165B3B1C21
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
** Programming Started **
Warn : no flash bank found for address 0x10000000
Warn : no flash bank found for address 0x10011f00
** Programming Finished **
** Verify Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x1000575e msp: 0x20041fa0
Error: checksum mismatch - attempting binary compare
embedded:startup.tcl:530: Error: ** Verify Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 591
at file "embedded:startup.tcl", line 530
*** [upload] Error 1`

I have not tried different target yet.

sbishop61, 2024-03-27:
Interestingly, after a reflash, it now sees all the flash:

`0.002 (  2) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
0.002 (  0) - (II)                      Welcome to Yet Another Picoprobe v1.20-f726bca
0.002 (  0) - (II) Features:
0.002 (  0) - (II)   [CMSIS: DAPv1] [CMSIS: DAPv2] [MSC: DAPLink] [CDC: target] [CDC: probe debug] [Net-NCM: SysView Echo IPerf]
0.002 (  0) - (II) Probe HW:
0.002 (  0) - (II)   Pico @ 240MHz (dual core)
0.002 (  0) - (II) IP:
0.002 (  0) - (II)   192.168.14.1
0.003 (  1) - (II) Compiler:
0.003 (  0) - (II)   clang 17.0.1 - release build
0.003 (  0) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
4.501 (...) - (II) FLASH: Copying custom flash code to 0x20010000 (620 bytes)
4.516 ( 15) - (II) SWD clk req   : 15000kHz = 240000kHz / (6 * (2 + 171/256)), eff : 14992kHz
4.526 ( 10) - (II)
4.526 (  0) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
4.526 (  0) - (II) Target vendor     : RaspberryPi
4.526 (  0) - (II) Target part       : RP2040
4.527 (  1) - (II) Flash             : 0x10000000..0x10ffffff (16384K)
4.527 (  0) - (II) RAM               : 0x20000000..0x2003ffff (256K)
4.527 (  0) - (II) SWD frequency     : 15000kHz
4.527 (  0) - (II) SWD max frequency : 25000kHz
4.528 (  1) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
4.528 (  0) - (II)
4.628 (100) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
5.216 (588) - (II) ---- no RTT_CB found`

rgrr, 2024-03-27:
hmmm... as far as I ca see, the conection to the target is ok, because the probe correctly detects the target.

Could you please install pyocd and check it with that?

ventZl, 2024-04-24:
I've had the same issue with older build of OpenOCD 0.11 (codebase patched by RPi foundation to support the original Picoprobe). All the wiring was OK (used previously with picoprobe firmware) and PyOCD worked out of the box. Upgrading to upstream OpenOCD 0.12 fixed the issue for me.

rgrr, 2025-02-27:
closing this, because no response

closed 2025-02-27 by rgrr