rgrr/yapicoprobe #110: Error connecting DP: cannot read IDR. Potential issue with stm32f0/similar chips

dexter93, 2024-06-25:
New YAPicoprobe ( Pico) out of the box, fresh connectors and the latest release bin ( [yapicoprobe-0120-pico-f726bca.uf2](https://github.com/rgrr/yapicoprobe/releases/download/v1.20/yapicoprobe-0120-pico-f726bca.uf2) )

When launching OpenOCD
`openocd -f interface/cmsis-dap.cfg -f target/stm32f0x.cfg`

it errors out with
```
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Error: Error connecting DP: cannot read IDR
```
enabling more debug print shows complete mayhem

```
Debug: 617 742 cmsis_dap.c:1100 cmsis_dap_swd_switch_seq(): JTAG-to-SWD
Debug: 618 742 cmsis_dap.c:872 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 619 744 cmsis_dap.c:1105 cmsis_dap_swd_switch_seq(): JTAG-to-DORMANT
Debug: 620 744 cmsis_dap.c:1120 cmsis_dap_swd_switch_seq(): DORMANT-to-SWD
Debug: 621 745 cmsis_dap.c:872 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Error: 622 746 adi_v5_swd.c:366 swd_connect_single(): Error connecting DP: cannot read IDR
Debug: 623 746 command.c:544 run_command(): Command 'dap init' failed with error code -4
User : 624 746 command.c:608 command_run_line(): 
Debug: 625 746 command.c:544 run_command(): Command 'init' failed with error code -4
User : 626 746 command.c:608 command_run_line(): 
Debug: 627 746 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Debug: 628 747 cmsis_dap.c:1110 cmsis_dap_swd_switch_seq(): SWD-to-JTAG
```

```
$  openocd -v
Open On-Chip Debugger 0.12.0
```

Status LED keeps blinking. Power levels are correct and there are no shorts in SWDIO/CLK - continuity also verified proper.

Is there something that needs attention on non-rp2040 debug targets? This seems like PEBKAC/misconfiguration

Thanks in advance!

rgrr, 2024-06-26:
Unfortunately I have no STM32 available, nevertheless some questions:

* what is the voltage level of your STM32F0?
* does it work with other probe software?
* could you try an old version (1.00) of yapicoprobe?  Sometime ago there was an issue with STMs due to tuning
* have you tried pyocd?

rgrr, 2025-02-27:
closing this, because no response

closed 2025-02-27 by rgrr