rgrr/yapicoprobe #111: Failed to install release firmwares

FabienTregan, 2024-07-04:
Hi,

  I have a pico clone board (the purple ones) that I can flash with the official picoprobe firmware without any problem. I want to try yapicoprobe (mainly to have RTT).

  Using the latest release firmware, the probe don't even enumerate. (nothing in `dmesg` output)

  Using the previous release, it probe-rs finds it but can't apparently communicate:

```
~/code/k-hz50$ probe-rs list
The following debug probes were found:
[0]: Debug Probe (CMSIS-DAP) -- 2e8a:000c:154A92160E464253 (CMSIS-DAP)

~/code/k-hz50$ probe-rs info
Probing target via JTAG

Error identifying target using protocol JTAG: The probe does not support the JTAG protocol.

Probing target via SWD

Error showing ARM chip information for Debug Port Default: An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Trying alternate multi-drop debug ports
Error showing ARM chip information for Debug Port Multidrop(16787751): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Error showing ARM chip information for Debug Port Multidrop(285223207): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

Output of same commands with the official debugprobe for pico firmware:

```
~/code/k-hz50$ probe-rs list
The following debug probes were found:
[0]: Debugprobe on Pico (CMSIS-DAP) -- 2e8a:000c:154A92160E464253 (CMSIS-DAP)

~/code/k-hz50$ probe-rs info
Probing target via JTAG

Error identifying target using protocol JTAG: The probe does not support the JTAG protocol.

Probing target via SWD

Error showing ARM chip information for Debug Port Default: An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Trying alternate multi-drop debug ports
ARM Chip with debug port Multidrop(1002927):
Debug Port: DPv2, MINDP, Designer: Raspberry Pi Trading Ltd, Part: 0x1002, Revision: 0x0, Instance: 0x00
└── 0 MemoryAP
    └── ROM Table (Class 1), Designer: ARM Ltd
        ├── Cortex-M0 SCS   (Generic IP component)
        │   └── CPUID
        │       ├── IMPLEMENTER: ARM Ltd
        │       ├── VARIANT: 0
        │       ├── PARTNO: Cortex-M0+
        │       └── REVISION: 1
        ├── Cortex-M0 DWT   (Generic IP component)
        └── Cortex-M0 BPU   (Generic IP component)

ARM Chip with debug port Multidrop(11002927):
Debug Port: DPv2, MINDP, Designer: Raspberry Pi Trading Ltd, Part: 0x1002, Revision: 0x0, Instance: 0x01
└── 0 MemoryAP
    └── ROM Table (Class 1), Designer: ARM Ltd
        ├── Cortex-M0 SCS   (Generic IP component)
        │   └── CPUID
        │       ├── IMPLEMENTER: ARM Ltd
        │       ├── VARIANT: 0
        │       ├── PARTNO: Cortex-M0+
        │       └── REVISION: 1
        ├── Cortex-M0 DWT   (Generic IP component)
        └── Cortex-M0 BPU   (Generic IP component)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

Any idea what I am doing wrong?

(I copied the udev config as said in the readme and rebooted)

thanks,
Fabien.

FabienTregan, 2024-07-04:
After building the current main commit (`13dd5d1`), it probe-rs list behaves as intended:

```
$ probe-rs info
Probing target via JTAG

Error identifying target using protocol JTAG: The probe does not support the JTAG protocol.

Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv2, MINDP, Designer: Raspberry Pi Trading Ltd, Part: 0x1002, Revision: 0x0, Instance: 0x00
└── 0 MemoryAP
    └── ROM Table (Class 1), Designer: ARM Ltd
        ├── Cortex-M0 SCS   (Generic IP component)
        │   └── CPUID
        │       ├── IMPLEMENTER: ARM Ltd
        │       ├── VARIANT: 0
        │       ├── PARTNO: Cortex-M0+
        │       └── REVISION: 1
        ├── Cortex-M0 DWT   (Generic IP component)
        └── Cortex-M0 BPU   (Generic IP component)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```
but even after forcing RTT on:
```
cmd: show        
5.319 (...) - (II) magic 0xfeedbabe
5.319 (  0) - (II)     name: config
5.319 (  0) - (II)     size: 17
5.319 (  0) - (II) ------------- device configuration
5.319 (  0) - [config]

5.320 (  1) - rtt=1

5.320 (  0) - (II) -------------
```
probe-rs still can't attach RTT:
```
warning: `k-hz50` (bin "k-hz50") generated 1 warning (run `cargo fix --bin "k-hz50"` to apply 1 suggestion)
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.04s
     Running `probe-rs run --chip RP2040 --protocol swd target/thumbv6m-none-eabi/debug/k-hz50`
      Erasing ✔ [00:00:00] [#############################################################################################] 16.00 KiB/16.00 KiB @ 120.55 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [##############################################################################################] 16.00 KiB/16.00 KiB @ 33.35 KiB/s (eta 0s )    Finished in 0.646s
Error: Failed to attach to RTT

Caused by:
    Error attempting to attach to RTT: RTT control block not found in target memory.
    - Make sure RTT is initialized on the target, AND that there are NO target breakpoints before RTT initialization.
    - For VSCode and probe-rs-debugger users, using `halt_after_reset:true` in your `launch.json` file will prevent RTT
    initialization from happening on time.
    - Depending on the target, sleep modes can interfere with RTT.
```

rgrr, 2024-07-04:
ok... seems like I have to check the release images.

For RTT you actually have to do nothing with yapicoprobe except attaching a terminal program to one of the probes VCOM ports.

rgrr, 2024-08-06:
Could you please try the latest release images?: https://github.com/rgrr/yapicoprobe/releases/tag/v1.21

rgrr, 2025-02-27:
closing this, because no response

closed 2025-02-27 by rgrr