rgrr/yapicoprobe #113: change rtt settings on target, rtt and systemview not work

PR2403, 2024-07-24:
In SEGGER_RTT_Conf.h
```
// Up-channel 0: RTT
// Up-channel 1: SystemView
//
#ifndef   SEGGER_RTT_MAX_NUM_UP_BUFFERS
  #define SEGGER_RTT_MAX_NUM_UP_BUFFERS             (3)     // Max. number of up-buffers (T->H) available on this target    (Default: 3)
#endif
//
// Most common case:
// Down-channel 0: RTT
// Down-channel 1: SystemView
//
#ifndef   SEGGER_RTT_MAX_NUM_DOWN_BUFFERS
  #define SEGGER_RTT_MAX_NUM_DOWN_BUFFERS           (3)     // Max. number of down-buffers (H->T) available on this target  (Default: 3)
#endif
```
default is 3, I change it to 2 for less ram useage
this cause rtt and systemview not work, no out put, channel reconizating error
```
9.618 (904) - (II) ---- RTT_CB found at 0x2000ff88
9.620 (  2) - (II)      rtt_check_channel_from_target: 0 20004DD4  1024     0    25
9.622 (  2) - (II)      rtt_check_channel_to_target  : 0 20004C14     8     0     0
9.623 (  1) - (II)      rtt_check_channel_from_target: 1 20004814  1024     0     0
 9.626 (  3) - (II) ---- RTT_CB at 0x2000ff88 seems to be inactive, searching again...
10.228 (602) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
11.132 (904) - (II) ---- RTT_CB found at 0x2000ff88
11.134 (  2) - (II)      rtt_check_channel_from_target: 0 20004DD4  1024     0    30
11.136 (  2) - (II)      rtt_check_channel_to_target  : 0 20004C14     8     0     0
11.137 (  1) - (II)      rtt_check_channel_from_target: 1 20004814  1024     0     0
11.140 (  3) - (II) ---- RTT_CB at 0x2000ff88 seems to be inactive, searching again...
11.742 (602) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
```
0x20004C14 is channel 1 down buffer address not channel 0

rgrr, 2024-07-25:
- how is the probe debug output, if you are using the default settings?
- how much RAM do you save?
- which RTT version are you using?

rgrr, 2024-07-25:
Ok... can confirm that there is something: config in Zephyr

    CONFIG_SEGGER_RTT_MAX_NUM_DOWN_BUFFERS=2
    CONFIG_SEGGER_RTT_MAX_NUM_UP_BUFFERS=3

is ok, but

    CONFIG_SEGGER_RTT_MAX_NUM_DOWN_BUFFERS=2
    CONFIG_SEGGER_RTT_MAX_NUM_UP_BUFFERS=2

is not.  Output is as follows for NOT OK:

    4.421 (  1) - (II) ---- RTT_CB found at 0x20004f10
    4.421 (  0) - (II)      rtt_check_channel_from_target: 0 2000808C  8192     0    62
    4.422 (  1) - (II)      rtt_check_channel_to_target  : 0 2000A170     8     0     0
    4.422 (  0) - (II)      rtt_check_channel_from_target: 1 2000A178 16384     0 16383

and for OK:

    4.416 (  0) - (II)      rtt_check_channel_from_target: 0 200080A4  8192     0    62
    4.417 (  1) - (II)      rtt_check_channel_to_target  : 0 20008094    16     0     0
    4.417 (  0) - (II)      rtt_check_channel_from_target: 1 2000A190 16384     0 16382
    4.418 (  1) - (II)      rtt_check_channel_to_target  : 1 2000A188     8     0     0

which means, that the second `rtt_check_channel_to_target` is not found.

I will look into this.

rgrr, 2024-08-06:
There is an obvious bug in rtt_io.c: number of aUp[] is not/wrong taken into account for access to aDown[].
Previous functionality is more or less luck.

PR #117 (2024-08-06): Reworking SEGGER_RTT_BUFFER access
closes #113
closed 2024-08-06 by rgrr