rgrr/yapicoprobe #123: RP2350 as probe target

rgrr, 2024-09-18:


holysnippet, 2024-10-05:
Is it normal for Flash programming and debugging not to work on the RP2350 as target?
I'm not impatient and I'm not here to complain, I'd just like to know if the problem is with my hardware?

rgrr, 2024-10-06:
yep, rp2350 as target is currently not working.  That's my current WIP ;-)  rp2350 as probe HW does hopefully work, but not fully tested.

If you want to help, you are welcome.

holysnippet, 2024-10-06:
Hello, thank you for your reply. I spent a good two hours trying to get my probe to work with a Pico 2 before coming to you with the question! I foolishly thought that the area was larger (from ~250k to ~500k) and that switching from one to the other was as simple as a new memory mapping for the probe.
I wrote a piece of code that was used to program Microchip PICs four or five years ago. But I don't know what it takes to program a Pico. I haven't even the slightest idea what kind of difference there is between the RP2040 and the RP2350 in terms of their Flash programming.
I don't think I can be of much help. But your probe helps me almost every day!

holysnippet, 2024-10-08:
Hi, I'm thinking of testing this later today, I may need it if I can finish switching my code from RP2040 to 2350. I intend to take advantage of the Pico 2's improvements, which seem to be able to calculate more efficiently. Thanks for all your hard work! The official picoprobe doesn't yet support the Pico 2 as a target, as I understand it. Impressive.

rgrr, 2024-10-08:
RP2350 as a target is under construction.  Actually the original picoprobe does not have to support the RP2350 directly, because it is just a simple DAP interface.  Yapicoprobe has to do some RP2350 handling, because of RTT and drag-and-drop support.

Currently only openocd (raspi fork) supports the rp2350, pyocd does unfortunately not.

PR #126 (2024-10-07): RP2350 as probe - fixes
THIS IS WORK IN PROGRESS

RP2350 as a target

closes #123
PR #127 (2024-10-21): RP2350 as a debug target plus RTT
WORK IN PROGRESS

closes #123
closed 2024-10-21 by rgrr