rgrr/yapicoprobe #131: [Newbie] Connection

maincraft-io, 2024-12-09:
```
/Users/web/Library/xPacks/@xpack-dev-tools/openocd/0.11.0-5.1/.content/bin/openocd \
    -c "gdb_port 50000" \
    -c "tcl_port 50001" \
    -c "telnet_port 50002" \
    -s /Users/web/Library/xPacks/@xpack-dev-tools/openocd/0.11.0-5.1/.content/scripts \
    -f /Users/web/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl \
    -f interface/cmsis-dap.cfg \
    -f target/rp2040-core0.cfg \
    -c "adapter speed 5000"
```

```
xPack OpenOCD arm64 Open On-Chip Debugger 0.11.0+dev (2022-09-01-20:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
CDLiveWatchSetup
cortex_m reset_config sysresetreq

adapter speed: 5000 kHz

Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4602C200B554639
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.2
Info : CMSIS-DAP: Serial# = E4602C200B554639
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Error: [rp2040.cpu] Could not find MEM-AP to control the core
Warn : target rp2040.cpu examination failed
Info : starting gdb server for rp2040.cpu on 50000
Info : Listening on port 50000 for gdb connections
```

1) I’m sharing the error logs. Could you please help me understand what I’m doing wrong?
`Error: [rp2040.cpu] Could not find MEM-AP to control the core`
2) yapicoprobe-0121-pico-11bf6c0.uf2 
3) original debugprobe_on_pico.uf2 working is ok

maincraft-io, 2024-12-09:
/Users/web/Library/xPacks/@xpack-dev-tools/openocd/0.11.0-5.1/.content/bin/openocd -f interface/cmsis-dap.cfg -f target/rp2040-core0.cfg -c "adapter speed 25000"    -c "program /Users/webad/Development/FD_BLOCK.elf verify reset; shutdown;"

```
xPack OpenOCD arm64 Open On-Chip Debugger 0.11.0+dev (2022-09-01-20:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
cortex_m reset_config sysresetreq

adapter speed: 25000 kHz

Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4602C200B554639
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.2
Info : CMSIS-DAP: Serial# = E4602C200B554639
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 25000 kHz
Info : SWD DPIDR 0x10212927
Error: [rp2040.cpu] Could not find MEM-AP to control the core
Warn : target rp2040.cpu examination failed
Info : starting gdb server for rp2040.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: [rp2040.cpu] Could not find MEM-AP to control the core
Error: [rp2040.cpu] Target not examined, reset NOT asserted!
embedded:startup.tcl:1435: Error: ** Unable to reset target **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 1472
at file "embedded:startup.tcl", line 1435

```

I didn't connect the reset pin. Is it absolutely required?

maincraft-io, 2024-12-09:
**Openocd 12.0.4**

`/Users/web/Library/xPacks/@xpack-dev-tools/openocd/0.12.0-4.1/.content/bin/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 25000"    -c "program /Users/web/Development/FD_BLOCK.elf verify reset; shutdown;"`

```
xPack Open On-Chip Debugger 0.12.0+dev-01685-gb9224c0c0-dirty (2024-08-02-22:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
adapter speed: 25000 kHz
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4602C200B554639
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.2
Info : CMSIS-DAP: Serial# = E4602C200B554639
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 25000 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core0] Examination succeed
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Examination succeed
Info : [rp2040.core0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to breakpoint, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
** Programming Started **
Info : Found flash device 'win w25q16jv' (ID 0x001540ef)
Info : RP2040 B0 Flash Probe: 2097152 bytes @0x10000000, in 32 sectors

Info : Padding image section 1 at 0x1001dbd0 with 48 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x1001dc00 .. 0x1001ffff
Error: CMSIS-DAP command mismatch. Expected 0x6 received 0x7e
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Error: Failed to read memory at 0x2001c5e4
Error: Could not allocate bounce buffer for flash programming. Can't continue
Error: error writing to flash at address 0x10000000 at offset 0x00000000
embedded:startup.tcl:1813: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 1878
at file "embedded:startup.tcl", line 1813

```

rgrr, 2024-12-09:
hmmm... could you use a newer version of openocd?

maincraft-io, 2024-12-09:
I reverted to the original firmware: 
https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.2.0. 
I didn’t change any wires; the connection is exactly the same. **yapicoprobe** not working for me :(

**Log Pico OpenOCD:**
```
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4602C200B554639
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core0] Examination succeed
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Examination succeed
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
```

rgrr, 2024-12-09:
There was also a somehow surprising issue, that openocd no longer works with "_DAP_PACKET_COUNT_OPENOCD   2".

If you can compile YaPicoprobe you can check it on your own.  Otherwise a will provide an experimental version with "_DAP_PACKET_COUNT_OPENOCD   1" later on. (give me some hours, still at work)

maincraft-io, 2024-12-09:
> There was also a somehow surprising issue, that openocd no longer works with "_DAP_PACKET_COUNT_OPENOCD 2".
> 
> If you can compile YaPicoprobe you can check it on your own. Otherwise a will provide an experimental version with "_DAP_PACKET_COUNT_OPENOCD 1" later on. (give me some hours, still at work)

Thank you for your prompt response! My current level of expertise isn’t sufficient to fully understand the situation as I’m a beginner. I’ll wait for your experimental update.

rgrr, 2024-12-09:
> I reverted to the original firmware: https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.2.0. I didn’t change any wires; the connection is exactly the same. Original is work.
> 
> **yapicoprobe** not working for me :(
> 
> **Log Pico OpenOCD:**
> 
> ```
> Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
...
> ```

Could you please provide the openocd 0.12 output for yapicoprobe?

maincraft-io, 2024-12-09:
> ...
>
>
> Could you please provide the openocd 0.12 output for yapicoprobe?

`webad@mb ~ % /Users/webad/.pico-sdk/openocd/0.12.0+dev/openocd -s /Users/webad/.pico-sdk/openocd/0.12.0+dev/scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c adapter speed 5000;    `
                                                                                     
```
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Unexpected command line argument: speed
webad@MacBook-Pro-Semen ~ % /Users/webad/.pico-sdk/openocd/0.12.0+dev/openocd -s /Users/webad/.pico-sdk/openocd/0.12.0+dev/scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg                   
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4602C200B554639
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.2
Info : CMSIS-DAP: Serial# = E4602C200B554639
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz
Warn : Too long SWD WAIT, issuing DAPABORT
Warn : Too long SWD WAIT, issuing DAPABORT
Warn : Too long SWD WAIT, issuing DAPABORT
Error: Failed to connect multidrop rp2040.dap0
```

maincraft-io, 2024-12-09:
Wires din't changed. RX/TX/VCC/GND/SWD
**Without RESET**

rgrr, 2024-12-09:
Okay... saw it that you already provided one above.  In the log above you gave a wrong cmd line argument.  Could you please test with 5MHz adapter speed?  Just as the first log, but with openocd 0.12

maincraft-io, 2024-12-09:
> Okay... saw it that you already provided one above. In the log above you gave a wrong cmd line argument. Could you please test with 5MHz adapter speed? Just as the first log, but with openocd 0.12

My operation system is OSX 15.1.1 (Apple Silicon arm64), macbook pro.
Thanks for your patience

`webad@mb ~ % /Users/webad/.pico-sdk/openocd/0.12.0+dev/openocd -s /Users/webad/.pico-sdk/openocd/0.12.0+dev/scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"`

```
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
adapter speed: 5000 kHz
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4602C200B554639
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.2
Info : CMSIS-DAP: Serial# = E4602C200B554639
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Warn : Too long SWD WAIT, issuing DAPABORT
Warn : Too long SWD WAIT, issuing DAPABORT
Warn : Too long SWD WAIT, issuing DAPABORT
Error: Failed to connect multidrop rp2040.dap0
```

maincraft-io, 2024-12-09:
```
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
adapter speed: 5000 kHz
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4602C200B554639
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.2
Info : CMSIS-DAP: Serial# = E4602C200B554639
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core0] Examination succeed
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Examination succeed
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
```

`Executing task: /Users/webad/.pico-sdk/openocd/0.12.0+dev/openocd -s /Users/webad/.pico-sdk/openocd/0.12.0+dev/scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c adapter speed 5000; program "/Users/webad/Development/ARC/Dev/FD_BLOCK/build/FD_BLOCK.elf" verify reset exit `

```
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4602C200B554639
Error: CMSIS-DAP command mismatch. Sent 0x0 received 0x1
Error: CMSIS-DAP command CMD_INFO failed.
** OpenOCD init failed **
shutdown command invoked
```

maincraft-io, 2024-12-09:
**Here’s an additional log; hopefully, it might help.**
```
---- Opened the serial port /dev/tty.usbmodem11108 ----
)), eff : 100kHz
1.648 (...) - (II) =================================== DAPv2 disconnect target
1.649 (  1) - (II) =================================== MSC connect target
1.771 (122) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.115 (115) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.115 (115) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.115 (115) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.115 (115) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.115 (115) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.115 (115) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.115 (115) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.115 (115) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== MSC connect target
0.114 (114) - (II) =================================== MSC disconnect target
0.000 (...) - (II) =================================== DAPv2 connect target, host OpenOCD with two big buffers, buffer: 2x512bytes
0.005 (  5) - (II) SWD clk req   : 4800kHz = 144000kHz / (6 * (5 + 0/256)), eff : 4800kHz
1.635 (...) - (II) =================================== DAPv2 disconnect target
0.000 (...) - (II) FLASH: Copying custom flash code to 0x20010000 (620 bytes)
0.015 ( 15) - (II) SWD clk req   : 15000kHz = 144000kHz / (6 * (1 + 154/256)), eff : 14985kHz
0.020 (  5) - (II) SWD clk req   : 14400kHz = 144000kHz / (6 * (1 + 171/256)), eff : 14388kHz
0.025 (  5) - (II) 
0.025 (  0) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
0.025 (  0) - (II) Target vendor     : RaspberryPi
0.025 (  0) - (II) Target part       : RP2040
0.026 (  1) - (II) Flash             : 0x10000000..0x101fffff (2048K)
0.026 (  0) - (II) RAM               : 0x20000000..0x2003ffff (256K)
0.026 (  0) - (II) SWD frequency     : 14400kHz
0.026 (  0) - (II) SWD max frequency : 25000kHz
0.026 (  0) - (II) ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
0.026 (  0) - (II) 
0.127 (101) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
0.827 (700) - (II) ---- no RTT_CB found
0.000 (...) - (II) =================================== DAPv2 connect target, host OpenOCD with two big buffers, buffer: 2x512bytes
0.005 (  5) - (II) SWD clk req   : 4800kHz = 144000kHz / (6 * (5 + 0/256)), eff : 4800kHz
0.000 (...) - (II) =================================== DAPv2 disconnect target
0.101 (101) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
1.724 (...) - (II) ---- no RTT_CB found
6.225 (...) - (II) =================================== DAPv2 connect target, host OpenOCD with two big buffers, buffer: 2x512bytes
0.000 (...) - (II) =================================== DAPv2 disconnect target
0.101 (101) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
1.724 (...) - (II) ---- no RTT_CB found
0.000 (...) - (II) =================================== DAPv2 connect target, host UNKNOWN, buffer: 1x64bytes
0.799 (799) - (II) =================================== DAPv2 disconnect target
0.900 (101) - (II) searching RTT_CB in 0x20000000..0x2003ffff, prev: 0x00000000
2.523 (...) - (II) ---- no RTT_CB found
0.000 (...) - (II) =================================== DAPv2 connect target, host UNKNOWN, buffer: 1x64bytes
```

maincraft-io, 2024-12-09:
Is this log enough?

maincraft-io, 2024-12-09:
[image]

[image]

maincraft-io, 2024-12-09:
> Give this a shot: https://github.com/rgrr/yapicoprobe/releases/tag/v1.23

**It works!**
**Amazing work!** you are genius! 

[image]
[image]

rgrr, 2024-12-09:
Good to hear that.  Thank you for your fast responses.

I'll close this and wish you happy hacking & debugging

closed 2024-12-09 by rgrr