rgrr/yapicoprobe #132: RP2040 RTT Support / Sample?

maincraft-io, 2024-12-09:
I’m trying to use RTT, but there’s no output in the console. What am I doing wrong?

```
#include "SEGGER_RTT.h"
#include "SEGGER_RTT_printf.c"

int main() 
{
    SEGGER_RTT_Init();
    
    while (true)
    {
        SEGGER_RTT_WriteString(0, "It works\n"); 
        sleep_ms(1000);
    }
}
```

[image]

rgrr, 2024-12-09:
RTT with the yapicoprobe does not appear via telnet 19111, instead it is using the other USB-CDC.

Don't know how to find that interface in MacOS, but you have found the one for probe debug output, so it should be a snap.

maincraft-io, 2024-12-09:
I will check. Thank you.

tao-j, 2024-12-11:
there are 3 cdc
1. target uart redirect
2. probe debug
3. rtt? (this issue is referring to)

However, the third cdc does not show up as a serial device either on linux or mac, have no windows to test. Does it show as a COM port in windows?

rgrr, 2024-12-11:
No, there are two CDC.  One which does UART/RTT redirect and one for probe debug.

If your log shows something like

```
---- no RTT_CB found
```

can should not expect anything from RTT.

rgrr, 2025-02-27:
closing this, because no response

closed 2025-02-27 by rgrr