rgrr/yapicoprobe #145: Wake sleeping CPU on RTT write?

esky-software, 2025-03-08:
Writing to RTT downlink when the CPU is sleeping in WFI does not wake the CPU.
In some circumstances this would be nice to have happen.

AFAICT waking the CPU requires a 'debug event', which background memory access is not.
Ideally such an event could be generated without halting the CPU, but I don't see any obvious way to do this.
So I guess just a halt & continue after the RTT write would be sufficient.

Adding this as an optional capability would be very handy.

rgrr, 2025-03-10:
Interesting idea.  One could also think about sending a debug event if the RTT up buffer is read via DAP.

In the "Armv8-M Architecture Reference Manual" one can read under "B2.1.1 The Wait for Event (WFE) instruction" about WFE wakeup events:
* ...
* If debug is enabled, a debug event.
* ...

If you continue reading about WFI wakeup, it is much more complex.

So I'm wondering why you intend to use WFI?

And also, what is the use case for your idea?  If you have a look at Seggers RTT read/write, you can find blocking operations which do an active wait.  TMO while the debug IF is active, current is no actual limitation.

esky-software, 2025-03-11:
My use case is that I'm using RTT as a debug terminal, which means I can't sleep indefinitely - I need to wake periodically to check for RTT data.

It's a minor annoyance, but one that can be easily fixed with this feature.

I agree that generating a debug event after uplink read also sounds good.  I'd suggest that these be enabled/disabled independently.

WFI vs WFE is an interesting question, especially since so much of their details are implementation-specific.  After combing the armv6m and armv8m manuals and the rp2040 and rp2350 datasheets and corralling all the snippets of information scattered through them, I've come to the conclusion that Arm's *intention* is for WFE to stop the processor but not much else, and WFI to also trigger low-power modes in surrounding subsystems.  But I've never found a satisfactory explanation of the whole deal.  (Probably not too hard to test for any specific processor though.)

BTW I'm currently working with rp2040.